STMicroelectronics/STM32CubeF0 #2: STEVAL-IDP004V1 (master-board) either not communicating to PC through RS485 or not detecting a sensor out of the box

pranavmondhe, 2019-08-06:
Hi,
We have recently purchased the STEVAL-BFA001V1B kit and planned to connect the STEVAL-IDP004V1 to the STEVAL-IDP005V1 as we planned to use the STEVAL-IDP004V1 GUI interface to allow for streaming off data from the sensor.

The manuals we have used are:

https://www.st.com/content/ccc/resource/sales_and_marketing/presentation/product_presentation/group0/47/a2/91/a0/fe/ca/42/89/steval-bfa001v1b_getting_started_guide/files/steval-bfa001v1b_quick_start_guide.pdf/jcr:content/translations/en.steval-bfa001v1b_quick_start_guide.pdf
https://www.avnet.com/opasdata/d120001/medias/docus/190/STEVAL-BFA001V1B%20User%20Manual.pdf
 
They make reference to updating the firmware of the STEVAL-IDP005V1 through dragging across "STEVAL-IDP005_ConditionMonitoring_IOL_v1.1.0.bin." The GUI after doing this results in the following error:
**********************************************************************************************
"[>> TX] START

[*ERR*] Command ERROR (TRASMISSION FAIL) - RECEIVED DATA 

[>> TX] COMMAND END

[*ERR*] Command ERROR (TRASMISSION FAIL) - RECEIVED DATA 

 

[*ERR*] Communication with STEVAL-IDP004V1 BOARD failed.

    Please reset the BOARD and the GUI and retry."
**********************************************************************************************
The 2nd manual refers to disconnect the assembly that we tried with no hope of any differences.

The STEVAL-IDP004V1 itself has 4 green flashes lights for each sensor (sensor 1, sensor 2 and so fourth) that has not changed implying that the sensor (STEVAL-IDP005V1) is not detected as a sensor.

Additionally, we are unable to determine whether there is any communication between the STEVAL-IDP004V1 and the PC.

We have uploaded another project (Condition_Monitoring_SRV) through the System Workbench by running/debugging and then connected through putty which following a reset of the STEVAL-IDP005V1 means data is streamed from the sensor.

Please help.

Thank you.

Best Regards
Pranav Mondhe

pranavmondhe, 2019-08-06:
Hi,
After changing the baud rate to 38400, GUI application works to some extend but fails on sensor command.
I am following the user manual but the baud rate mentioned in it is completely different.!!!

*********************************************************************************************
[>> TX] START
[<< RX]                                               *********************************
                                              *       STMicroelectronics      *
                                              * IO - Link MULTI-PORT SOLUTION *
                                              *********************************

SELECT MASTER OR DEVICE

[>> TX] MASTER
[<< RX] 
INSERT ADDRESS IC

[>> TX] 0
[<< RX] 
INSERT OPERATING MODE

[>> TX] WR_S
[<< RX] 
INSERT REGISTER VALUE

[>> TX] 096,248,033,122,122,122,122,
[<< RX] 
PROGRAMMING DONE

[>> TX] COMMAND END
[<< RX] 
SELECT MASTER OR DEVICE

[>> TX] DEVICE
[<< RX] 
INSERT SLAVE NODE

[>> TX] 0
[<< RX] 
INSERT SENSOR COMMAND

[>> TX] IDS
[*ERR*] Command ERROR (TRASMISSION FAIL) - RECEIVED DATA 

[>> TX] COMMAND END
[*ERR*] Command ERROR (TRASMISSION FAIL) - RECEIVED DATA I=I5R?
[*ERR*] IDS COMMAND FAIL - NODE 0 - ATTEMPT 1
[*ERR*] Communication with STEVAL-IDP004V1 BOARD failed.
       Please reset the BOARD and the GUI and retry.
************************************************************************************************

ALABSTM, 2019-08-09:
Dear Pranav,

We are very pleased with the interest you show to our products. In case you found an issue in the user manual, it is always worth reporting it. It helps us enhance the quality of our documents and we do thank you for that.

However, this public organization on GitHub is intended to promote co-development and issues posted here shall exclusively be related to the software published within the different repositories.

For any other issue (e.g. related to documentation, to hardware, to ecosystem, etc.), you shall address your questions to the [ST Community](https://community.st.com/s/topiccatalog). There you shall look for the adequate topic and submit your question or remark.

As for the issue you posted here, allow us to close it now. Thank you for your understanding.

We hope you keep on enjoying using our products and look forward to receiving further issues and pull-request from you (once pull-requests become accepted).

With regards,

pranavmondhe, 2019-08-09:
@ALABSTM 
Thanks for looking at my issue.
I think the documentation which i was referring is totally wrong. Its been more than 10 days, i haven't got any reply from ST community family to my requests.

Main issues are :
1. Documentation for STEVAL-UK1001V1 are completely misguided for deploying .bib/hex into the sensor IDP005V1.
2. In the firmware examples of STEVAL-IDP004V1, there is not mention about support of IDP005V1 sensor (in its release notes its saying to use IDP003V1)
3. There is not mention to upgrade firmware of STEVAL-IDP004V1. Does it needs to?
4. GUI program mentioned inside Utility folder for STEVAL-UK1001V1 firmware - doesn't supports Baud rate of 230400 but it supports 38400.

Can you please forward my notest to developement/support team?

Best Regards
Pranav

ALABSTM, 2019-08-09:
Dear Pranav,

I will try to. Hopefully you get the answers you are expecting to these setbacks you are facing.

With regards,

pranavmondhe, 2019-08-12:
@ALABSTM 
Okay-. Thank you.

Kind Regards
Pranav Mondhe

pranavmondhe, 2019-08-15:
@ALABSTM 
I have not heard back from anyone from support team about my issue.

Best Regards
Pranav

CCASTM, 2019-08-19:
Hello Pranav,

Ali is currently out of office. Let me try to help you.
This product has been developed based on a STM32F0 microcontroller, to insure the promotion of the L6360 IC. 
As previously explained by Ali, we, as STM32 Introduction Package team, are in charge of providing regular deliveries and support on GitHub for the 13 STM32 series drivers and generic project software.  But you can understand that we cannot support all projects and products developed in every ST divisions based on our STM32 references. You need to contact the ST provider directly.

As proposed by Ali, you can ask your question to the ST Community forum (https://community.st.com/s/) 
An alternative solution is to use ST OLS (On-Line support) channel. 
-> Go to the STEVAL-IDP004V1 page  (https://www.st.com/en/evaluation-tools/steval-idp004v1.html)
-> From the "Support and Community" select "Support Home" then "Support Request -via web form" . You will be invited to subscribe to myST and then you will get access to a specific form that will drive you to the ST Division FAEs 

Hope this will help you
Regards
Christophe

pranavmondhe, 2019-08-19:
Hello Christophe
Thanks for your reply.
I have just created a new request on https://www.st.com/en/evaluation-tools/steval-idp004v1.html and my case id is : 00086892

Please help.
Thanks & Regards
Pranav Mondhe

pranavmondhe, 2019-08-21:
Hello Christophe
Are we going to create a ticket/issue to solve my problem here?
Please help.

Best Regards
Pranav Mondhe

CCASTM, 2019-08-21:
Hello Pranav.
We will not take the responsibility of the problem that you have submitted. As I indicated, it is the team that has developed the board and its related software. 
The reference of this support case is then the On-Line-Support ID that you have created. 
Regards
Christophe

pranavmondhe, 2019-08-28:
Okay- thanks.

Kind Regards
Pranav

ALABSTM, 2019-09-09:
Hi Pranav,

As this issue is now handled at On-Line Support level, allow me to close it. Thank you for your comprehension.

With regards,

closed 2019-09-09 by ALABSTM