STMicroelectronics/STM32CubeF0 #4: UART ISR becomes interrupt bound due to incomplete implemenation of UART Recieve Timeout

david-oz00, 2020-03-29:
**Describe the set-up**
* Hardware: Proprietary hardware containing STM32F030RC
* IDE: VisualGDB with GCC 9.2.1
* HAL: 1.11.0

**Describe the bug**
When the UART receive timeout is enabled, it is possible the ISR executes without clearing the interrupt.  Consequently, the ISR is continuously called using 100% of the microprocessor time.

**How To Reproduce**
1. Indicate the global behavior of your application project.
A serial connection which uses intercharacter timeout to indicate the end of message.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
HAL_UART_IRQHandler() ISR function in stm32f0xx_hal_uart.c

3. The use case that generates the problem.
A serial connection that uses intercharacter timeout to detect 'end of message'.  The receive timeout is enabled using the following functions:

```
HAL_UART_ReceiverTimeout_Config(mHuart, mBitsPerCharacter);
HAL_UART_EnableReceiverTimeout(mHuart);

SET_BIT(mHuart->Instance->CR1, USART_CR1_RTOIE);  // See Note

```
Note: The receive timeout interrupt has to be enabled manually

4. How we can reproduce the problem.
You need to generate a receive timeout interrupt and ensure no other UART interrupts are pending.  Following this condition the ISR will become interrupt bound.

**Additional context**

The defect is in the ISR function HAL_UART_IRQHandler().  The following code:

```
if ((errorflags != 0U)
      && (((cr3its & USART_CR3_EIE) != 0U)
          || ((cr1its & (USART_CR1_RXNEIE | USART_CR1_PEIE)) != 0U)))

```
should be:

```
if ((errorflags != 0U)
      && (((cr3its & USART_CR3_EIE) != 0U)
          || ((cr1its & (USART_CR1_RXNEIE | USART_CR1_PEIE | USART_CR1_RTOIE)) != 0U)))

```
Notice **USART_CR1_RTOIE** is not present in the original code.

When this is fixed, can you also integrate the enabling and disabling of the receive timeout interrupt into the HAL so it doesn't have to be done manually. 

**Screenshots**
None

ASELSTM, 2020-03-31:
Hi David-oz00,

Thank you for reporting this issue. It will be forwarded to our development team to check its relevance. You will be updated as soon as the provide us with feedback. Thank you once more.

With regards,

ASELSTM, 2020-04-01:
Hi David-oz00,

Our development teams confirmed the issue you pointed out. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-04-01:
ST Internal Reference: 83742

ASELSTM, 2020-12-09:
Hi @david-oz00,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.11.2** of the STM32CubeF0 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM