STMicroelectronics/STM32CubeF0 #5: Fault in HAL_CEC code

newbrain, 2020-05-19:
**Problem Location:**
stm32f0xx_hal_cec.c file in ./Drivers/STM32F0xx_HAL_Driver/Src.

**Problem:**
The _HAL_CEC_Transmit_IT()_ function, in conjunction with the _HAL_CEC_IRQHandler()_ ISR will always append an extra byte to the end of message, unless only one byte (header only: CEC ping) is sent.
This will cause many TVs to ignore the whole message (some might work, e.g. some recent Sony).
The problem has been confirmed with TVs from Sony, LG, Samsung and other brands, with models of varied vintage and checked with an oscilloscope.
The proposed solution has been verified in the same way.

**Note: This is the third time I report this fault, that affects also the libraries for other STM32 models with CEC, e.g. STM32F7.**
First report: TECH037361 or CTNbg19058 (2017)
Second report: Case# 00082897 (2019)
Both times the fault was acknowledged (as was the proposed solution) and said to be corrected in a next version.

**Root Cause:**
When a single byte (i.e. a CEC header) is to be transmitted, HAL_CEC_Transmit_IT() correctly sets the EOM bit:

```
    /* in case of no payload (Size = 0), sender is only pinging the system;
       Set TX End of Message (TXEOM) bit, must be set before writing data to TXDR */
    if (Size == 0U)
    {
      __HAL_CEC_LAST_BYTE_TX_SET(hcec);
    }

```

HAL_CEC_IRQHandler() will then just clean up and invoke the TX complete callback.
When any payload needs to be transmitted (Size > 0), its size will be assigned to TxXferCount:

```
  /* initialize the number of bytes to send,
   * 0 means only one header is sent (ping operation) */
  hcec->TxXferCount = Size;
```

In the IRS, when a TXBR interrupt is being served and the last byte is about to be sent, TxXferCount will still be 1:

```
  /* CEC TX byte request interrupt ------------------------------------------------*/
  if((reg & CEC_FLAG_TXBR) != RESET) 
  {
    if (hcec->TxXferCount == 0U)
    {
      /* if this is the last byte transmission, set TX End of Message (TXEOM) bit */
      __HAL_CEC_LAST_BYTE_TX_SET(hcec);
      hcec->Instance->TXDR = *hcec->pTxBuffPtr;
      hcec->pTxBuffPtr++;
    }
    else
    {    
      hcec->Instance->TXDR = *hcec->pTxBuffPtr;
      hcec->pTxBuffPtr++;
      hcec->TxXferCount--;
    }  
    /* clear Tx-Byte request flag */
    __HAL_CEC_CLEAR_FLAG(hcec,CEC_FLAG_TXBR); 
  } 
```

The check for TxXferCount is the source of the error:
when the last byte is going to be transmitted, TxXferCount is still 1, not 0, as it's decremented only after writing TXDR; for this reason, the TXBR interrupt will be invoked one additional time (possibly reading past the end of the transmit buffer!) and an unknown value will be appended to the message.

**Solution:**
Correct the check for the last message byte in HAL_CEC_IRQHandler():

```
  /* CEC TX byte request interrupt ------------------------------------------------*/
  if((reg & CEC_FLAG_TXBR) != RESET) 
  {
    if (--hcec->TxXferCount == 0U) /* pre-decrement the size before the comparison */
    {
      /* if this is the last byte transmission, set TX End of Message (TXEOM) bit */
      __HAL_CEC_LAST_BYTE_TX_SET(hcec);
    }
    /* In all cases transmit the byte */
    hcec->Instance->TXDR = *hcec->pTxBuffPtr++;
    /* clear Tx-Byte request flag */
    __HAL_CEC_CLEAR_FLAG(hcec,CEC_FLAG_TXBR); 
  } 
```

**How to reporduce:**
Just send any CEC message with a payload.

ASELSTM, 2020-05-20:
Hi @newbrain,

Thank you for this report. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

ASELSTM, 2020-05-27:
Hi @newbrain,

As promised we have reported your report to our development teams and they confirmed the issue. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-05-27:
ST Internal Reference: 87112

ASELSTM, 2020-12-09:
Hi @newbrain,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.11.2** of the STM32CubeF0 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2020-12-09 by ASELSTM