STMicroelectronics/STM32CubeF0 #7: HAL I2C OA1EN always set even when slave is not used

eelmor, 2020-11-02:
Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_i2c.c

`/*---------------------------- I2Cx OAR1 Configuration ---------------------*/
  /* Disable Own Address1 before set the Own Address1 configuration */
  hi2c->Instance->OAR1 &= ~I2C_OAR1_OA1EN;

  /* Configure I2Cx: Own Address1 and ack own address1 mode */
  if (hi2c->Init.AddressingMode == I2C_ADDRESSINGMODE_7BIT)
  {
    hi2c->Instance->OAR1 = (I2C_OAR1_OA1EN | hi2c->Init.OwnAddress1);
  }
  else /* I2C_ADDRESSINGMODE_10BIT */
  {
    hi2c->Instance->OAR1 = (I2C_OAR1_OA1EN | I2C_OAR1_OA1MODE | hi2c->Init.OwnAddress1);`

I2C_OAR1_OA1EN is always set in HAL_I2C_Init which is causing issues in multi-master situations. The I2C driver will address match (address 0 if not configured) even when only used in Master-mode causing a stall on the bus if not handled separately in user-code.

ASELSTM, 2020-11-03:
Hi @eelmor,

Thank you for your report. It has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

Best regards,

ASELSTM, 2021-06-09:
Hi @eelmor,

Sorry for the delay in coming back to you.

Your request has been forwarded to our development team and they are asking for more clarification about the multi-master use.  Please allow me then to ask you few questions : 

- As the Address “0” is reserved to general call, please make sure that you have configured a correct (real) address.

- Are you using our MCU as master with another MCU (2 MCUs used) ?  Or are you using only one MCU and declaring different I2C instances as masters within your application ? 

With regards,

eelmor, 2021-06-10:
Hello @ASELSTM,

1. I'm not sure about the exact matching criteria as I don't know the exact data being sent by the other master, but the I2C_ISR ADDR bit keeps getting set when using the I2C peripheral in Master mode which stalls the peripheral. I don't see any reason to set the OA1EN bit when used as Master only.
2. Yes, two masters on the I2C bus (2x MCU).

ASELSTM, 2021-06-11:
Hi @eelmor,

Referring to your feedback, the behavior you are detecting means that you didn't set an own address value on the initialization phase. Indeed, an own address should be set since that in the initialization phase the I2C don't know if it is configured as master or slave.

Even if you use the device as a master,  you need to set an own address since that the I2C is set as slave by default.  The I2C becomes master once the start bit is generated and becomes slave back after a sop generation as mentioned  in the [reference manual](https://www.st.com/resource/en/reference_manual/dm00031936-stm32f0x1stm32f0x2stm32f0x8-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) within section 

[image]

With regards,

ASELSTM, 2021-06-11:
Hi @eelmor,

Referring to your feedback, the behavior you are detecting means that you didn't set an own address value on the initialization phase. Indeed, an own address should be set since that in the initialization phase the I2C don't know if it is configured as master or slave.

Even if you use the device as a master,  you need to set an own address since that the I2C is set as slave by default.  The I2C becomes master once the start bit is generated and becomes slave back after a sop generation as mentioned  in the [reference manual](https://www.st.com/resource/en/reference_manual/dm00031936-stm32f0x1stm32f0x2stm32f0x8-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) within section **"26.4.4 Mode selection"**.

[image]

With regards,

eelmor, 2021-06-11:
Hello @ASELSTM ,

If I'm not setting a slave address, HAL should not set the OA1EN bit.

ASELSTM, 2021-06-17:
Hi @eelmor,

As mentioned above, the STM32 I2C instances are by default configured as slave unless they are used as master by calling the master mode APIs.  So that, at initialization phase, since the I2C is set as slave mode during this phase, an address is necessary to allow the I2C to recognize it self. 

To resume, the I2C instances are operating in slave mode unless a start condition is generated so that setting a slave address is primordial and mandatory.  Even if you are not setting any address in your source code, the general call address **0x00** is taken by default and OA1EN will be set anyway.

In order to allow a correct use of the I2C, a correct address (different from general call address `0x00`) should be set as slave address.

Hope this would help you.

With regards,

eelmor, 2021-06-28:
Hello @ASELSTM,

It doesn't help me. How can I initialize I2C in HAL without slave mode active? Right now I need workarounds to do this. I don't want it to respond to any address.

BR,
Jon

ASELSTM, 2021-07-09:
Hi @eelmor,

It is actually not possible as the I2C is designed to operate in salve mode unless it generates a START condition. From an IP design point of view, the I2C is always configured as slave by default.

In multi-master mode there must be a sequence between the applications to manage the call to the HAL API in synchronization. Thus, Master 1 call Master_Transmit, until end then Master 2 call  Master_Transmit until end. Please make sure that you are using the right sequence within your application.

Please allow me then to close this thread. Thank you for your comprehension.

With regards,

closed 2021-07-09 by ASELSTM

eelmor, 2021-07-09:
All that's needed is for OA1EN to stay cleared during HAL Init when I don't specify a slave address and it works correctly. I want to be able to do this without hacking HAL. How can you say that it's not possible?