STMicroelectronics/STM32CubeF0 #8: cdc Tc complete callback

diefx, 2020-11-13:
https://github.com/STMicroelectronics/STM32CubeF0/blob/7b44872c87d4ef209386bb6bebe821f4b526b576/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L694

Hello

Is it possible to add a callback function when a transmition  has been completed?, just like the callback in the usart library, it will be very useful to add some extra code without the need to modify the library, like for instance i want to release a semaphore every time i finish sending a message trough the usb

jrahlf, 2020-12-23:
This was added to some devices, but it seems like they forgot to update the F0. 
See here for the F4:
https://github.com/STMicroelectronics/STM32CubeF4/blob/b95577991ef8891c62c41e0845903e182cdf60d3/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c#L697

ALABSTM, 2021-01-19:
Hi @diefx,

Thank you for asking. The callback has been implemented in the frame of version **v2.6.0** of the USB Device library. As pointed out by @jrahlf, this version of the library is supported by version **v1.25.0** STM32CubeF4 and newer.

Regarding the STM32CubeF0 firmware, current version is **v1.11.2** which supports version **v2.5.3** of the USB Device library. Unfortunately, there is no plan for the moment to upgrade to the newer one. I will check further with our development teams and let you know.

In the meanwhile, you could use the **v2.6.0** from the STM32CubeF4. There is no dependency wit HAL. However, pay attention that you may face **compatibility break** at application-level (whether with ready-to-use applications provided in the firmware package or with the Cube MX-generated ones).

I hope this helps. Do not hesitate if you have other questions related.

With regards,

ALABSTM, 2021-07-01:
Hi @diefx,

I hope you are fine. As said before, the STM32CubeF0 firmware, current version is **v1.11.2** which supports version **v2.5.3** of the USB Device library. Unfortunately, there is **no plan** to upgrade to a newer one.

However, you could use the **latest** version available on the [stm32_mw_usb_device](https://github.com/STMicroelectronics/stm32_mw_usb_device) repository, which is fully compatible with **v2.5.3**.

Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2021-07-01 by ALABSTM