STMicroelectronics/STM32CubeF0 #11: Overflow error when compiling with GCC 10.1

rgilton, 2021-02-03:
**Describe the set-up**
 * Compiling for STM32F042F6
 * gcc 10.1

**Describe the bug**
When compiling the file `Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c`, GCC 10.1 flags a warning about an overflow.

~~~
$ arm-none-eabi-gcc -o /tmp/foo.o -c -Wall -Werror -DSTM32F042 -DSTM32F042F6 -DSTM32F042Fx -DSTM32F042x6 -DSTM32F042xx -DSTM32F04x -DSTM32F04xxx -DSTM32F0xx -DSTM32F0xxxx -DSTM32Fxxx -DSTM32Fxxxxx -mcpu=cortex-m0 -mlittle-endian -mthumb -mfloat-abi=soft -Iinc -Ibuild/inc -Ilibs/stm32cubef0/Drivers/CMSIS/Device/ST/STM32F0xx/Include -Ilibs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Inc -Ilibs/stm32cubef0/Drivers/CMSIS/Include libs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c
In file included from libs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_hal_def.h:30,
                 from libs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_hal_rcc.h:29,
                 from inc/stm32f0xx_hal_conf.h:182,
                 from libs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_hal.h:30,
                 from libs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c:46:
libs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c: In function 'HAL_CRC_DeInit':
libs/stm32cubef0/Drivers/CMSIS/Device/ST/STM32F0xx/Include/stm32f0xx.h:198:41: error: overflow in conversion from 'int' to 'uint8_t' {aka 'volatile unsigned char'} changes value from '(int)hcrc->Instance->IDR & -256' to '0' [-Werror=overflow]
  198 | #define CLEAR_BIT(REG, BIT)   ((REG) &= ~(BIT))
      |                                         ^
libs/stm32cubef0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c:205:3: note: in expansion of macro 'CLEAR_BIT'
  205 |   CLEAR_BIT(hcrc->Instance->IDR, CRC_IDR_IDR);
      |   ^~~~~~~~~
cc1: all warnings being treated as errors
~~~

RKOUSTM, 2021-02-09:
Hi @rgilton,

Thank you for contribution. The point you raised is already known and concerns the compatibility of our HAL APIs with the volatile strictness in C++20. This issue has also been previously reported by another user ([link](https://github.com/STMicroelectronics/STM32CubeWB/issues/22)).

This is one of the main points in the tasks queue of our development teams. We cannot share a date regarding its deployment and publication. We will keep you informed. 

Thank you for you again for your contribution.

With regards

RKOUSTM, 2021-05-10:
Duplicate of [STM32CubeWB#22](https://github.com/STMicroelectronics/STM32CubeWB/issues/22).

RKOUSTM, 2021-05-19:
ST Internal Reference: 107263

RJMSTM, 2023-11-22:
Hello @rgilton ,

Thank you for your contribution, Unfortunately, our development team has declined your proposal. According to them, no intention indeed for Cube1 to be compliant with C++20.

see (https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2021/p2327r0.pdf)
Now idea yet of the outcome

In this case, please allow me to close this issue. Thank you for your comprehension and thank you again for your proposal.

With regards,
Rania

closed 2023-11-22 by RJMSTM