STMicroelectronics/STM32CubeF0 #12: Including `stm32f0xx_ll_adc.h` introduces ` `-Wunused-parameter` warnings into client code

mayl, 2021-03-08:
The offending section is below:

https://github.com/STMicroelectronics/STM32CubeF0/blob/4390ff6bfb693104cf97192f98c3dc9e3a7c296a/Drivers/STM32F0xx_HAL_Driver/Inc/stm32f0xx_ll_adc.h#L1411-L1415

The parameter `Register` is not used.  However, because this is a function body defined in a header file, any _users_ of this library wind up hitting compiler warnings for this oversight.  I build my projects with `-Werror`, and so this becomes an error for me.  I don't know for sure what the project's preferred fix is but a couple that come to mind (some could work in combination):

* Don't define the body of this function in the header file
* Eliminate this unnecessary `Register` argument from the function definition
* Add a `(void) Register;` line to the function to appease the compiler and eliminate the warning

RKOUSTM, 2021-03-08:
Hi @mayl,

Thank you for your contribution.

Actually, the point you raised has already been raised in another issue [#12](https://github.com/STMicroelectronics/STM32CubeF3/issues/12). The issue you pointed out has been confirmed, a fix we will be implemented and made available in future release. 

Thank you once again for your contribution.

with regards,

RKOUSTM, 2021-03-08:
ST Internal Reference: 102615

RKOUSTM, 2021-08-16:
Hi @mayl,

The fix you requested has been implemented and is now available in the frame of the latest STM32CubeF0 FW package V1.11.3 release. 

This issue can be closed now. Thank you again for your contribution.

With regards,

closed 2021-08-16 by RKOUSTM

mayl, 2021-08-28:
Thanks RKOUSTRM and STmicro for fixing this!