STMicroelectronics/STM32CubeF0 #14: Missing ";" in cmsis_os.h, FreeRTOS - CMSIS V1.11.x

therocketelectrician, 2021-11-18:
**Describe the set-up**
Programming a STM32F072B
With STM32CubeIDE
Version: 1.7.0
Build: 10852_20210715_0634 (UTC)
and
STM32CubeMX - STM32 Device Configuration Tool
Version: 6.3.0-RC5
Build: 20210714-1111 (UTC)

**Describe the bug**
I get this build error:
`C:/Users/User/STM32Cube/Repository/STM32Cube_FW_F0_V1.11.3/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.h:825:1: error: expected ';' before 'extern'
  825 | extern osMailQDef_t os_mailQ_def_##name`

**How To Reproduce**
1.1 Create a FreeRTOS project for a STM32F072B with the FreeRTOS API-CMSIS V1
1.2 Create an `osMailQDef(writeAnalog_mail, 2, analogData_t);` in the main.h
1.3 In `main.c`, put that code:
```
#define osObjectsExternal
#include "main.h"
main(){
     writeAnalog_mail = osMailCreate(osMailQ(writeAnalog_mail), NULL);
}
```

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
In the file name is : "cmsis_os.h".
````
//  ==== Mail Queue Management Functions ====

#if (defined (osFeature_MailQ)  &&  (osFeature_MailQ != 0))     // Mail Queues available

/// \brief Create a Mail Queue Definition.
/// \param         name          name of the queue
/// \param         queue_sz      maximum number of messages in queue
/// \param         type          data type of a single message element
/// \note CAN BE CHANGED: The parameter to \b osMailQDef shall be consistent but the
///       macro body is implementation specific in every CMSIS-RTOS.
#if defined (osObjectsExternal)  // object is external
#define osMailQDef(name, queue_sz, type) \
extern struct os_mailQ_cb *os_mailQ_cb_##name \
extern osMailQDef_t os_mailQ_def_##name
#else                            // define the object
#define osMailQDef(name, queue_sz, type) \
struct os_mailQ_cb *os_mailQ_cb_##name \
const osMailQDef_t os_mailQ_def_##name =  \
{ (queue_sz), sizeof (type), (&os_mailQ_cb_##name) }
#endif 
````

3. The use case that generates the problem.
Write what is explained in "1" and simply build. An error should appear. The way to solve it, which worked for me was to add ';' at the end of `extern struct os_mailQ_cb *os_mailQ_cb_##name`, `extern osMailQDef_t os_mailQ_def_##name`, `struct os_mailQ_cb *os_mailQ_cb_##name`, `onst osMailQDef_t os_mailQ_def_##name = { (queue_sz), sizeof (type), (&os_mailQ_cb_##name) `,
The end result should look like this:
````
#if defined (osObjectsExternal)  // object is external
#define osMailQDef(name, queue_sz, type) \
extern struct os_mailQ_cb *os_mailQ_cb_##name; \
extern osMailQDef_t os_mailQ_def_##name;
#else                            // define the object
#define osMailQDef(name, queue_sz, type) \
struct os_mailQ_cb *os_mailQ_cb_##name; \
const osMailQDef_t os_mailQ_def_##name =  \
{ (queue_sz), sizeof (type), (&os_mailQ_cb_##name); }
#endif 
````

**Additional context**
Maybe look and test all "extern" definition if there is any ";" missing.

ALABSTM, 2022-01-30:
Hi @therocketelectrician,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

RKOUSTM, 2022-02-25:
Hi @therocketelectrician,

First we would like to thank you for your report.

We tried to reproduce the issue you have described using the STM32CubeMX v6.5.A13 and the IAR v8.50.6 running on STM32F072RB board, with no success. I really hope you could find the root cause of this problem and, ideally, solve it.

Regarding the point for which you opened the issue, as it is irrelevant, allow me to close the issue. Anyway, if you ever become able to reproduce systematically the faulty behavior and find out that the problem is related to our drivers, do not hesitate to recontract me. We could reopen this issue again.

Thank you for your comprehension and thank you again for your report. We are looking forward to reading from you again.

With regards,

closed 2022-02-25 by RKOUSTM