STMicroelectronics/STM32CubeF0 #15: missing static in HAL_RCC_GetSysClockFreq()

CanastraRF, 2022-01-03:
Hi
In the function 

`uint32_t HAL_RCC_GetSysClockFreq(void)`

are two const array 

```
  uint8_t const aPLLMULFactorTable[16U] = { 2U,  3U,  4U,  5U,  6U,  7U,  8U,  9U,
                                         10U, 11U, 12U, 13U, 14U, 15U, 16U, 16U};
  uint8_t const aPredivFactorTable[16U] = { 1U, 2U,  3U,  4U,  5U,  6U,  7U,  8U,
                                           9U,10U, 11U, 12U, 13U, 14U, 15U, 16U};

```
they should bee static

Please fix this

RKOUSTM, 2022-01-07:
Hi @CanastraRF,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you again for you report.

With regards,

RKOUSTM, 2022-01-07:
ST Internal Reference: 120497

closed 2023-04-10 by ALABSTM