STMicroelectronics/STM32CubeF0 #16: SystemCoreClockUpdate() wrong value for STM32F030xC running PLL

pavel-kucera-ing, 2022-02-07:
**Setup**

Predefined macro STM32F030xC, using STM32F030CC MCU.

**Bug description**

When running this code after reset

```
RCC->CR |= RCC_CR_HSION;
while((RCC->CR & RCC_CR_HSIRDY) == 0U){
}
if((RCC->CFGR & RCC_CFGR_SWS) == RCC_CFGR_SWS_PLL)
{
    RCC->CFGR &= ~RCC_CFGR_SW;
    while((RCC->CFGR & RCC_CFGR_SWS) != RCC_CFGR_SWS_HSI){
    }
}
RCC->CR &= ~RCC_CR_PLLON;
while((RCC->CR & RCC_CR_PLLRDY) != 0U){
}
RCC->CFGR = ((RCC->CFGR & ~RCC_CFGR_PLLMUL) | RCC_CFGR_PLLMUL6);
RCC->CR |= RCC_CR_PLLON;
while((RCC->CR & RCC_CR_PLLRDY) == 0U){
}
FLASH->ACR |= 1U; // set latency to one wait state
RCC->CFGR |= RCC_CFGR_SW_PLL;
while((RCC->CFGR & RCC_CFGR_SWS) != RCC_CFGR_SWS_PLL){
}
SystemCoreClockUpdate();
```

SystemCoreClock variable holds value of 48000000.

**Awaited behavior**
SystemCoreClock should hold value of 24000000, as the RM0360 (DocID025023 Rev 4) in 7.4.2, page 102 states Bit 16 PLLSRC 0: HSI/2 selected as PLL input clock. Note the "/2". Verified behavior (before and after applying path below) on real HW.

**Correction patch**
```
diff --git Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/system_stm32f0xx.c Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/system_stm32f0xx.c
index 4761a85..c167609 100644
--- Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/system_stm32f0xx.c
+++ Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/system_stm32f0xx.c
@@ -211,7 +211,7 @@ void SystemCoreClockUpdate (void)
  || defined(STM32F078xx) || defined(STM32F071xB)  || defined(STM32F072xB) \
  || defined(STM32F070xB) || defined(STM32F091xC) || defined(STM32F098xx)  || defined(STM32F030xC)
         /* HSI used as PLL clock source : SystemCoreClock = HSI/PREDIV * PLLMUL */
-        SystemCoreClock = (HSI_VALUE/predivfactor) * pllmull;
+        SystemCoreClock = ((HSI_VALUE >> 1)/predivfactor) * pllmull;
 #else
         /* HSI used as PLL clock source : SystemCoreClock = HSI/2 * PLLMUL */
         SystemCoreClock = (HSI_VALUE >> 1) * pllmull;
```

Pavel Kucera

ASELSTM, 2022-02-16:
ST Internal Reference: 122887

ASELSTM, 2023-04-06:
Hi @pavel-kucera-ing,

According to the reference manual **RM0360** page 91, the STM32F030xC supports `HSI/PREDIV` so the proposed fix is not aligned with the specification. 
As this issue is invalid, please allow me thus to close this thread. Thank you fir your comprehension. 

With regards,

closed 2023-04-06 by ASELSTM