STMicroelectronics/STM32CubeF0 #19: I2C Slave address generate wrong?

cch9897, 2023-03-13:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * STM32F072CB
 * STM32CubeMX&gcc-arm-none-eabi-10.3-2021.10

**Describe the bug**
STM32CubeMX generate wrong i2c slave address.

**How To Reproduce**
1. Indicate the global behavior of your application project.
Use MCU be an i2c slave device.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
I2C 

3. The use case that generates the problem.
Use CubeMX generate i2c slave initial code,and the address convert not correct.

4. How we can reproduce the problem.
enable any i2c and fill slave address , then generate code. you will see the address convert wrong.

**Additional context**
The example 7bit address is 0x31 convert to 8bit is 0x62 but it seem fill 62 to the initial code?

**Screenshots**
[image]

HBOSTM, 2023-03-21:
Hello @cch9897 ,

Thank you for your contribution.
I don't agree with you on this point, as slave address equals 31 (DEC), if this value converted into 8 bits (shifted left by 1 bit) becomes 62 (DEC), 0x3E (HEX). So, please allow me to close this issue.

Best Regards,

closed 2023-03-22 by cch9897

cch9897, 2023-03-22:
Sorry for submitting the issue without verifying the I2C slave address first. I mistakenly entered it as hex. My apologies for any inconvenience caused.