STMicroelectronics/STM32CubeF0 #20: Unused parameter warnings in USB Middleware

mmerlo, 2025-02-04:
There are unused parameter warnings in the USB Middleware (see below). 

**Describe the set-up**
Firmware version: V1.11.5
Compiler: arm-none-eabi-gcc (15:13.2.rel1-2) 13.2.1 20231009
Compiler warning options: -Wall -Wextra -Wpedantic

**Warnings**
```
[build] /workspaces/src/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c: In function 'USBD_CtlError':
[build] /workspaces/src/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c:853:42: warning: unused parameter 'req' [-Wunused-parameter]
[build]   853 |                    USBD_SetupReqTypedef *req)
[build]       |                    ~~~~~~~~~~~~~~~~~~~~~~^~~
```
```
[build] /workspaces/src/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c: In function 'USBD_CDC_Init':
[build] /workspaces/src/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c:471:65: warning: unused parameter 'cfgidx' [-Wunused-parameter]
[build]   471 | static uint8_t  USBD_CDC_Init(USBD_HandleTypeDef *pdev, uint8_t cfgidx)
[build]       |                                                         ~~~~~~~~^~~~~~
```
```
[build] /workspaces/src/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c: In function 'USBD_CDC_DeInit':
[build] /workspaces/src/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c:549:67: warning: unused parameter 'cfgidx' [-Wunused-parameter]
[build]   549 | static uint8_t  USBD_CDC_DeInit(USBD_HandleTypeDef *pdev, uint8_t cfgidx)
[build]       |                                                           ~~~~~~~~^~~~~~
```

ALABSTM, 2025-02-05:
Hi @@mmerlo,

Thank you for having reported. Actually, this issue has already been fixed since version **2.6.0** of the library, as you can see [here](https://github.com/STMicroelectronics/stm32-mw-usb-device/blob/0b06460a43c30670a764cb7e58c5a7918e789f6e/Core/Src/usbd_ctlreq.c#L981).

Currently, the latest version released of the **USB Device library** is **2.11.3**. However, the **STM32CubeF0** firmware still supports version **2.5.3**.

I will check and let you know whether an upgrade is planned. In the meanwhile, you can get the latest version of the library from the [stm32-mw-usb-device](https://github.com/STMicroelectronics/stm32-mw-usb-device) repository and integrate it into your application.

With regards,

mmerlo, 2025-02-05:
Hi @ALABSTM,

Thank you for the reply. I had searched through all the STM32CubeFx repos but I was not aware of the [stm32-mw-usb-device](https://github.com/STMicroelectronics/stm32-mw-usb-device) repo. 

I am a bit confused on how the ST repos are organized and how I can find the latest version of the various codebases for the STM32F0. 

For example, the [STM32CubeF0](https://github.com/STMicroelectronics/STM32CubeF0) repo has a submodule in the `Drivers` folder to [stm32f0xx-hal-driver](https://github.com/STMicroelectronics/stm32f0xx-hal-driver). In this case, I would go to this submodule repo to look for the latest version. But, in [STM32CubeF0](https://github.com/STMicroelectronics/STM32CubeF0), the USB middleware is part of this repo and not a submodule and therefore I was unaware that I should go to [stm32-mw-usb-device](https://github.com/STMicroelectronics/stm32-mw-usb-device) to find the latest version.

Do you have any suggestions on how to best navigate these repos and find the current code? That should help a lot and help with submitting issues that are already fixed.

ALABSTM, 2025-02-06:
Hi @mmerlo,

The deployment of submodules is ongoing and the next step is to extend it to the middleware libraries. However, even though the USB Device library is replaced by a submodule, this does not mean it will automatically point on the latest version. A version upgrade of the library must happen only in the frame of a new release of the firmware.

Regarding a good way to navigate the repos, I suggest you look into the **README.md** file of the [STM32Cube_MCU_Overall_Offer](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer) repo, where all repositories related to the STM32 **MCU** offer are listed.

Please note similar repositories are also available in the **pinned** section on top of the organization's main page, which are related to the STM32 **MPU** and **AI** offers.

One last hint to keep in mind and that could help understanding how repos are managed. Along with the STM32CubeF0 firmware, to take this one as an example, the different **components** it is made of (**i.e.**, CMSIS Device interfaces, HAL-LL drivers, BSP drivers, and MW libraries) are also published in **dedicated** repositories.  Only projects (**i.e.**, demos, applications and examples) are embedded within the firmware repository.

This proved to be very useful for users only interested in those components, not in the whole firmware. This also allows us to deploy the submodules.

I hope I could answer your questions. Please, do not hesitate otherwise.

With regards,

ALABSTM, 2025-02-10:
Hi @mmerlo,

According to our development teams, there is currently **no plan** to upgrade the USB Device library supported by the STM32CubeF0 firmware. Please allow me to close this issue. I will let you know, should there be any plan update. Thank you for your comprehension.

With regards,

closed 2025-02-10 by ALABSTM

mmerlo, 2025-02-10:
Hi @ALABSTM,

Thank you for the thorough explanation on the repo organization and for letting me know the future plans of the dev team.

Am I correct in assuming the USB Device library still supports the STM32F0 MCUs and that the dev team has just decided to not upgrade this library in the Cube firmware?

ALABSTM, 2025-05-07:
Hi @mmerlo,

The USB Device library remains compatible with the STM32CubeF0 firmware. You will have to integrate it manually in your case.

With regards,